ArduCAM/RaspberryPi #2: ArduCAM UV-544 c++ control

FlYnN91, 2019-02-15:
Hi,

I have the ArduCAM UV-544 Rev B board connected to the CSI ribbon and I2C signals. I am using raspberry Pi 3B board. I would like to capture images via c++ application, but the examples are in Python.

I've seen the SPI_Camera folder with the demo application but if I build it and run it I get `SPI1 interface Error!` message in command line. 

Is my camera suitable for that application? I would like to autofocus and capture image using c++.

Any advise would be appreciated.

Thanks

ArduCAM, 2019-02-17:
@FlYnN91 Please check the folder : https://github.com/ArduCAM/RaspberryPi/tree/master/Motorized_Focus_Camera 
which is for motorized focus camera module. and we only provide python version code right now.
Code in SPI_Camera is for another camera not this motorized focus camera.

FlYnN91, 2019-02-22:
Sorry for late reply. I looked at the python example code for the motorized camera and it seems to be utilising SPI comms to talk to the camera. Could you provide the SPI communications protocol allowing me to port the python examples to C++?

I guess my only option is to get the USB version of the ArduCAM which has the C++ library. Is that correct?

ArduCAM, 2019-02-24:
Not we don't use SPI, we use the I2C, the protocol is the standard I2C protocol.
Since you got UC-544 pi camera board, it is nothing to do with SPI and USB, just go ahead to test the camera using the code in the folder: https://github.com/ArduCAM/RaspberryPi/tree/master/Motorized_Focus_Camera

FlYnN91, 2019-02-26:
Sorry, I did mean I2C. I would like to communicate with your camera from my C++ application so I cannot use the python scripts. Could you share the I2C protocol/messages so I can write a driver for it?

ArduCAM, 2019-02-27:
You can find the controller from : www.arducam.com/downloads/modules/DW9714-P8.pdf and figure out the I2C protocol.

FlYnN91, 2019-02-27:
That's perfect, thanks a lot.

closed 2019-02-27 by FlYnN91