ArduCAM/RaspberryPi #3: I2C device not detected

FlYnN91, 2019-03-18:
Hi,

I've been using the Autofocus UV-544 and the I2C ID 0x0c has stopped appearing/being detected. I can see that the ID 0x64 is still being found and the images still can be captured with raspistill, but the autofocus does not seem to work as the 0x0c ID is not responding. 

I2C configuration has not been changed, but it is very strange that this one ID to control the autofocus just does not respond any more.

Have you seen this kind of issue before? Any advice would be appreciated.

Thank you.

FarooqTanveer-Github, 2020-01-29:
I have the same issue. Did you find any solution?

FarooqTanveer-Github, 2020-01-29:
Add this line at the end of "sudo nano /boot/config.txt"
dtparam=i2c0=on

mariachiarabello, 2020-03-06:
> Add this line at the end of "sudo nano /boot/config.txt"
> dtparam=i2c0=on

Thank you very very much @FarooqTanveer-Github, it solved our problem!

aeleos, 2023-01-13:
For anyone else with this issue, if you are on bullseye you may need this instead:
dtparam=i2c10=on