ArduCAM/RaspberryPi #4: GPL violation - MIPI-Camera drivers include chunks of mmal-vchiq.c

6by9, 2019-03-26:
I'm happy that you're releasing products for the Pi, however I suspect you are breaching the GPL licence in providing only prebuilt kernel modules for your OV7251, OV9281, and IMX135 drivers.

A couple of minutes with a hex editor show they include various functions from [mmal-vchiq.c](https://github.com/raspberrypi/linux/blob/rpi-4.14.y/drivers/staging/vc04_services/bcm2835-camera/mmal-vchiq.c), which is licenced under GPLv2, and therefore requires that you provide source code to rebuild the modules.
Indeed the module itself declares itself to be licenced under the GPL licence. The module parameters are a giveaway as to the origin of the code too.
```
filename:       /home/pi/tmp/RaspberryPi/MIPI_Camera/ov7251-release/4.14.71-v7+/arducam-ov7251-v4l2-pi3.ko
version:        0.0.1
license:        GPL
author:         Everyone
description:    Broadcom 2835 MMAL video capture
srcversion:     5C76648EC6F250C09749E17
alias:          of:N*T*Carducam,ov7251C*
alias:          of:N*T*Carducam,ov7251
depends:        videobuf2-v4l2,videodev,videobuf2-vmalloc,videobuf2-core,v4l2-common
staging:        Y
name:           arducam_ov7251_v4l2
vermagic:       4.14.71-v7+ SMP mod_unload modversions ARMv7 p2v8 
parm:           debug:int
parm:           bcm2835_v4l2_debug:Debug level 0-2
parm:           video_nr:videoX start numbers, -1 is autodetect (array of int)
parm:           max_video_width:Threshold for video mode (int)
parm:           max_video_height:Threshold for video mode (int)
```
You're likely to hit issues as soon as the kernel goes to 4.19 due to mmal-vchiq having been broken out into an independent kernel module.

Can I point out that there is a [mainline kernel driver for ov7251](https://github.com/torvalds/linux/blob/master/drivers/media/i2c/ov7251.c). There is a kernel driver for the [Pi CSI-2 receiver peripheral](https://github.com/raspberrypi/linux/blob/rpi-4.19.y/drivers/media/platform/bcm2835/bcm2835-unicam.c), and a couple of example [dtoverlays](https://github.com/raspberrypi/linux/blob/rpi-4.19.y/arch/arm/boot/dts/overlays/ov5647-overlay.dts) for configuring it.
This is the better way of supporting these devices, and I'm happy to offer some support to you should you wish to go this route. If implemented cleanly, changes would be acceptable to be merged into the Pi kernel, and that would solve the issues of you having to rebuild against each kernel release.

ArduCAM, 2019-03-26:
Hi 6by9
Thank you very much for pointing out the licensing issue. We have removed the link to driver. 
We want to contact you shortly to discuss your suggestions on how to move forward.
Can you share with your email address by sending to admin@arducam.com?
Thanks in advance.
Regards
Arducam team

admin@arducam.com
 
From: 6by9
Date: 2019-03-26 18:54
To: ArduCAM/RaspberryPi
CC: Subscribed
Subject: [ArduCAM/RaspberryPi] GPL violation - MIPI-Camera drivers include chunks of mmal-vchiq.c (#4)
I'm happy that you're releasing products for the Pi, however I suspect you are breaching the GPL licence in providing only prebuilt kernel modules for your OV7251, OV9281, and IMX135 drivers.
A couple of minutes with a hex editor show they include various functions from mmal-vchiq.c, which is licenced under GPLv2, and therefore requires that you provide source code to rebuild the modules.
Indeed the module itself declares itself to be licenced under the GPL licence. The module parameters are a giveaway as to the origin of the code too.
filename:       /home/pi/tmp/RaspberryPi/MIPI_Camera/ov7251-release/4.14.71-v7+/arducam-ov7251-v4l2-pi3.ko
version:        0.0.1
license:        GPL
author:         Everyone
description:    Broadcom 2835 MMAL video capture
srcversion:     5C76648EC6F250C09749E17
alias:          of:N*T*Carducam,ov7251C*
alias:          of:N*T*Carducam,ov7251
depends:        videobuf2-v4l2,videodev,videobuf2-vmalloc,videobuf2-core,v4l2-common
staging:        Y
name:           arducam_ov7251_v4l2
vermagic:       4.14.71-v7+ SMP mod_unload modversions ARMv7 p2v8 
parm:           debug:int
parm:           bcm2835_v4l2_debug:Debug level 0-2
parm:           video_nr:videoX start numbers, -1 is autodetect (array of int)
parm:           max_video_width:Threshold for video mode (int)
parm:           max_video_height:Threshold for video mode (int)

You're likely to hit issues as soon as the kernel goes to 4.19 due to mmal-vchiq having been broken out into an independent kernel module.
Can I point out that there is a mainline kernel driver for ov7251. There is a kernel driver for the Pi CSI-2 receiver peripheral, and a couple of example dtoverlays for configuring it.
This is the better way of supporting these devices, and I'm happy to offer some support to you should you wish to go this route. If implemented cleanly, changes would be acceptable to be merged into the Pi kernel, and that would solve the issues of you having to rebuild against each kernel release.

anbello, 2019-05-20:
I would like to know if there is any news on this front.
I bought an arducam with OV9281 for raspberry pi ([this](https://www.uctronics.com/index.php/camera-modules/camera-for-raspberry-pi/mipi-camera-module-raspberr-pi/arducam-ov9281-mipi-camera-module-1mp-monochrome-global-shutter-camera.html) model) but actually I cannot use for lack of driver.

Thanks in advance
Andrea Belloni