ArduCAM/RaspberryPi #5: raspivid gst-launch performance

user1321, 2019-04-06:
Good Day!
Can you please share information about your product.
As I understood new ArduCam 2.1 is not compatible with raspivid, is it correct?
Instead we have to use /dev/video0 to get raw video (please, correct if I am wrong)
In the end too get compressed video we have to use gst-launch to compress it.
Did you check overall performance:
1. CPU load
2. Video latency comparing to raspivid

Am I wrong in that?
Thank you.

ArduCAM, 2019-04-06:
I'm not sure what is your Arducam 2.1 referring to, can you send me more information.

user1321, 2019-04-06:
Hi, 
Yes, sorry.
http://blog.uctronics.com/arducam-v2-1-multi-camera-adapter-board-raspberry-pi-8mp-camera-imx219/

ArduCAM, 2019-04-07:
IT works with raspistill and raspivid software and streaming the compressed video as well as gstreamer.
The overall performance is the same as offical V2 pi camera except the improvements on lenses.

user1321, 2019-04-07:
Got it. Thank you for your reply.

closed 2019-04-07 by user1321