ArduCAM/RaspberryPi #6: Where can I buy it?

FavorMylikes, 2019-04-08:
Could you give me a link for shopping?

ArduCAM, 2019-04-08:
You can buy from uctronics.com  they ship worldwide.

FavorMylikes, 2019-04-08:
And....

Which type then?

MIPI camera?

Hermann-SW, 2019-04-09:
https://www.uctronics.com/index.php/catalogsearch/result/?q=raspberry+mipi

FavorMylikes, 2019-04-09:
Thank you~

But the last thing that I wanna make sure is

All of those camera could programming motorized?

FavorMylikes, 2019-04-09:
Found it~

Thank you~

FavorMylikes, 2019-04-09:
Here the list, for those people who have same question
[Motorized Focus 8MP Camera IMX219 Sensor for Raspberry Pi](https://www.uctronics.com/index.php/camera-modules/camera-for-raspberry-pi/motorized-focus-8mp-camera-imx219-sensor-for-raspberry-pi.html)
[Motorized Focus 5MP Camera OV5647 Sensor for Raspberry Pi](https://www.uctronics.com/index.php/camera-modules/camera-for-raspberry-pi/motorized-focus-5mp-camera-ov5647-sensor-for-raspberry-pi.html)
[Arducam IMX135 MIPI 13MP Color Camera Module for Raspberry Pi](https://www.uctronics.com/index.php/camera-modules/camera-for-raspberry-pi/arducam-imx135-mipi-13mp-color-camera-module.html)

ArduCAM, 2019-04-09:
all these cameras lens focus can be controlled  by program.

zachjacobs, 2019-04-11:
Any chance that you can use the 13MP version with your multi-camera adapter board?
https://www.uctronics.com/index.php/multi-camera-adapter-module-for-raspberry-pi-board-8mp.html

ArduCAM, 2019-04-11:
we will test that and get back to you.