ArduCAM/RaspberryPi #7: Failed to read from camera

zachjacobs, 2019-05-03:
Hello,
I have purchased 2x motorized autofocus cameras from utronics.  I can't seem to get either of them to work.  I have enabled the camera and i2c interfaces through raspi-config as well as tested different cameras and Raspberry Pis.  I always get the same message:

> mmal: Cannot read camera info, keeping the defaults for OV5647
> mmal: mmal_vc_component_create: failed to create component 'vc.ril.camera' (1:ENOMEM)
> mmal: mmal_component_create_core: could not create component 'vc.ril.camera' (1)
> mmal: Failed to create camera component
> mmal: main: Failed to create camera component
> mmal: Camera is not detected. Please check carefully the camera module is installed correctly
> 
> Exception in thread Thread-1 (most likely raised during interpreter shutdown):⏎ 
>

zachjacobs, 2019-05-03:
If I plug a 5MP RPi camera into the camera interface it works without issues.

zachjacobs, 2019-05-03:
I can read the i2c device address on the camera module:
[image]

zachjacobs, 2019-05-03:
How am I even supposed to use this module?  The camera driver section in the documentation [here](http://www.arducam.com/downloads/modules/RaspberryPi_camera/IMX135_MIPI_Camera_Module_DS_v0.pdf) is rather difficult to complete.  I cannot seem to find either of the two scripts that it references in the auto install section:  `./preprocess.sh` and `./loaddriver.sh`.  I also can't find the `/boot/overlays/arducam-IMX135.dtbo` file listed in manual install section.  This device tree file doesn't exist in the official RPi image:  https://github.com/raspberrypi/firmware/tree/master/boot/overlays

ArduCAM, 2019-05-05:
Please contact info@arducam.com for more help.

zachjacobs, 2019-05-11:
@ArduCAM Hello, I have contacted them 5 days ago and I have not heard back.  Please help.  Do you support your products?

ArduCAM, 2019-05-11:
@zachjacobs we didn't get your email from info@arducam.com can you double check for that?

ArduCAM, 2019-05-11:
We neither found Zach nor Jacobs in our email inbox.

zachjacobs, 2019-05-13:
I used my work email zach@faradinesystems.com.

[image]

AlyonaMinina, 2019-05-15:
Hejsan Zach,

did you manage to sort out this problem? I ran into exactly the same issue yesterday with my brand new Arducam IMX219

AlyonaMinina, 2019-05-15:
Problem solved!

Got the issue fixed in no time with the help from arducam support team. It turned out to be a loose connector for the tiny brown cable that had to be clicked in properly.

[image]

UberEclectic, 2019-05-18:
How is it that the hardware is out, but the software featured prominently on the advertising blog and video is not posted anywhere here on GitHub under the Arducam account?
There are virtually no instructions for building and installing the driver, but I am experienced, so that much was a non-issue.  But no matter what I there is no reference to preprocess.sh or loaddriver.sh.

Crucially, I don't have any /dev/video/* handles.

So why aren't these new products even listed on the main ArduCam website?
Why do the buried product pages for them have blank documentation and software pages.

Please just post the proper code, here on GitHub?

ArduCAM, 2019-07-15:
New user guide can refer to this document, it applies to all MIPI camera modules.
www.arducam.com/downloads/modules/RaspberryPi_camera/B0163-13MP-IMX135-MIPI-Camera-Module.pdf

Gru-Preet, 2019-08-13:
How to use USB camera in Raspberry Pi 3 B +

I get the error when i try to take a snap:
pi@RPi:~ $ raspistill -o image.jpg
mmal: Cannot read camera info, keeping the defaults for OV5647
mmal: mmal_vc_component_create: failed to create component 'vc.ril.camera' (1:ENOMEM)
mmal: mmal_component_create_core: could not create component 'vc.ril.camera' (1)
mmal: Failed to create camera component
mmal: main: Failed to create camera component
mmal: Camera is not detected. Please check carefully the camera module is installed correctly