ArduCAM/RaspberryPi #11: MultiCamera for Raspberrypi v2.1 is not detected as video0

lrnzo, 2019-07-30:
hello there,

when i run the init_camera.sh, i get

----Detecting the /dev/video0 device
Can't find video0 device, please check the hardware connection! 

hardware connection is definetly ok, i check several times. when i try raspistill -o test.jpeg i get

mmal: Cannot read camera info, keeping the defaults for OV5647
mmal: mmal_vc_component_create: failed to create component 'vc.ril.camera' (1:ENOMEM)
mmal: mmal_component_create_core: could not create component 'vc.ril.camera' (1)
mmal: Failed to create camera component
mmal: main: Failed to create camera component
mmal: Camera is not detected. Please check carefully the camera module is installed correctly

when i plug in a single camera instead of the MultiCamera Adapter, it works normal, so the pi software/firmware/hardware seems to be ok. any ideas or any hints to debug this?

kind regards

lrnzo, 2019-07-30:
i should add this was on a raspberry 3 b+ with latest raspbian (buster) and kernel 4.19.58 see [here](https://pastebin.com/y7FyhS0L)  for output of raspinfo

ahmed-shariff, 2019-08-03:
Same exact issue here. I tried running `i2cdetect`, this is what I get. I have no clue what the output means, but here it is:

```
pi@raspberrypi:~ $ i2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- 04 05 -- 07 08 -- 0a -- 0c -- -- -- 
10: -- 11 -- -- -- 15 16 -- 18 19 1a -- 1c 1d -- -- 
20: 20 21 22 -- -- 25 -- -- 28 -- 2a 2b 2c -- 2e 2f 
30: -- 31 -- 33 -- -- -- 37 38 39 3a 3b 3c -- 3e -- 
40: -- -- 42 -- 44 -- 46 -- -- 49 -- 4b -- 4d -- 4f 
50: -- 51 -- -- 54 55 56 57 -- 59 -- -- -- -- 5e 5f 
60: -- -- -- -- -- 65 66 67 -- 69 -- -- 6c -- 6e -- 
70: 70 71 72 73 74 -- 76 77                         
pi@raspberrypi:~ $ i2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- 06 -- 08 -- -- -- -- 0d 0e 0f 
10: 10 -- 12 13 -- -- -- 17 -- 19 -- 1b 1c 1d -- 1f 
20: -- -- 22 -- 24 -- -- 27 -- 29 -- -- -- 2d -- 2f 
30: -- -- 32 33 34 -- -- 37 -- 39 3a 3b -- -- 3e 3f 
40: -- -- -- -- -- -- 46 -- -- 49 4a 4b -- 4d 4e -- 
50: 50 -- -- -- -- -- -- -- -- 59 5a 5b 5c 5d -- -- 
60: 60 -- -- -- -- -- 66 67 68 -- 6a 6b 6c -- 6e -- 
70: -- 71 72 -- 74 -- -- 77                         
pi@raspberrypi:~ $ i2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          03 04 05 06 07 -- -- 0a 0b -- 0d 0e 0f 
10: 10 -- -- 13 14 15 16 -- -- 19 -- -- -- 1d -- 1f 
20: -- -- 22 23 -- 25 -- 27 -- -- -- 2b 2c 2d 2e 2f 
30: -- 31 32 33 34 35 -- 37 38 39 3a 3b 3c -- 3e 3f 
40: -- -- 42 43 44 -- 46 -- -- -- 4a -- 4c -- -- -- 
50: -- -- 52 53 -- 55 56 57 -- -- 5a 5b 5c 5d 5e 5f 
60: -- -- -- -- 64 65 -- 67 68 -- 6a -- 6c 6d 6e -- 
70: 70 71 72 -- -- 75 -- 77                         
pi@raspberrypi:~ $ i2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          03 04 05 06 -- -- -- -- 0b 0c -- 0e 0f 
10: -- -- 12 -- 14 -- 16 -- 18 19 1a 1b -- -- -- -- 
20: -- -- 22 23 24 -- 26 -- -- -- -- -- 2c 2d -- 2f 
30: 30 -- -- 33 34 -- 36 37 38 -- -- -- 3c 3d 3e -- 
40: 40 41 -- 43 -- 45 46 47 -- -- -- 4b -- -- 4e 4f 
50: 50 51 52 -- -- -- 56 57 58 59 -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- 6a -- 6c 6d -- 6f 
70: -- -- 72 -- 74 -- 76 --
```

UCTRONICS, 2019-08-03:
For the new official raspbian firmware, the old Multi-Cam Board Adapter board is not compatible with it. Please visit this link to get new version board 
https://www.amazon.com/Arducam-Camera-Adapter-Raspberry-Cameras/dp/B07TYL36MC 
For old version, please use the firmware whose version is less than 9.8 including 9.8. You can run the lsb_release -a command to check your firmware version.

ahmed-shariff, 2019-08-04:
@UCTRONICS I actually have the version you have attached. My output of the `lsb_release -a`:
```
No LSB modules are available.
Distributor ID:	Raspbian
Description:	Raspbian GNU/Linux 10 (buster)
Release:	10
Codename:	buster
```

lrnzo, 2019-08-04:
@UCTRONICS i am having the problem with the current version of the board (2.1)

UCTRONICS, 2019-08-05:
@lrnzo @ahmed-shariff 
Please attach me the hardware setup diagram. I have tested the board just now and it can work normally.

ahmed-shariff, 2019-08-05:
I am not sure what you are referring to as the hardware setup diagram. Is it how I have wired the components together?

UCTRONICS, 2019-08-05:
@ahmed-shariff 
Yes. Please send me some detail picture for your hardware connection

ahmed-shariff, 2019-08-05:
[image]
[image]
[image]
[image]

@UCTRONICS does this help?

lrnzo, 2019-08-05:
i do not know i this helps or if this is what you mean, but i took photos of my setup:
https://cloud.ffnw.de/s/ZzRpEwDmS7x96JJ
https://cloud.ffnw.de/s/Gd5QFGWNFMXZiGJ

by the way it does also not work with raspbian stretch:
```
root@raspberrypi:~/RaspberryPi/...Adapter_V2.1# lsb_release -a
No LSB modules are available.
Distributor ID:	Raspbian
Description:	Raspbian GNU/Linux 9.9 (stretch)
Release:	9.9
Codename:	stretch
root@raspberrypi:~/RaspberryPi/...Adapter_V2.1# ./init_camera.sh 
----Test i2c1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- -- 
10: 10 -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- 64 -- -- -- -- -- -- -- -- -- -- -- 
70: 70 -- -- -- -- -- -- --                         
----Configure Adapter Board V2.1
Adapter Board V2.1 configure OK
----install v4l2 driver
v4l2 driver install OK
----Detecting the /dev/video0 device
Can't find video0 device, please check the hardware connection! 
```

lrnzo, 2019-08-05:
@ahmed-shariff i think your flex cable from rpis CSI connector to the board is in the wrong way. take a close look at how the contacts inside the sockets are and decide for yourself :)

UCTRONICS, 2019-08-05:
@lrnzo 
Thanks for your detail message, the firmware you are using is v9.9. The adapter board you are using is the old version. Please use our new version Adapter board V2.1.

lrnzo, 2019-08-05:
the board **has** v2.1 printed on it. on the backside it has UC-475 Rev.B.
https://cloud.ffnw.de/s/84brTLZT6JHzAAT
https://cloud.ffnw.de/s/PYtpBtMgNTSzBaW

lrnzo, 2019-08-05:
i am wondering wether i have to configure my gpios in some undocumented way? here is my output of `raspi-gpio get`. does this look correct for you?
```
root@raspberrypi:~# raspi-gpio get
BANK0 (GPIO 0 to 27):
GPIO 0: level=1 fsel=0 func=INPUT
GPIO 1: level=1 fsel=0 func=INPUT
GPIO 2: level=1 fsel=4 alt=0 func=SDA1
GPIO 3: level=1 fsel=4 alt=0 func=SCL1
GPIO 4: level=1 fsel=1 func=OUTPUT
GPIO 5: level=1 fsel=0 func=INPUT
GPIO 6: level=1 fsel=0 func=INPUT
GPIO 7: level=1 fsel=0 func=INPUT
GPIO 8: level=1 fsel=0 func=INPUT
GPIO 9: level=1 fsel=1 func=OUTPUT
GPIO 10: level=0 fsel=0 func=INPUT
GPIO 11: level=0 fsel=0 func=INPUT
GPIO 12: level=0 fsel=0 func=INPUT
GPIO 13: level=0 fsel=0 func=INPUT
GPIO 14: level=0 fsel=0 func=INPUT
GPIO 15: level=1 fsel=0 func=INPUT
GPIO 16: level=0 fsel=0 func=INPUT
GPIO 17: level=1 fsel=1 func=OUTPUT
GPIO 18: level=0 fsel=1 func=OUTPUT
GPIO 19: level=0 fsel=0 func=INPUT
GPIO 20: level=0 fsel=0 func=INPUT
GPIO 21: level=0 fsel=0 func=INPUT
GPIO 22: level=1 fsel=1 func=OUTPUT
GPIO 23: level=1 fsel=1 func=OUTPUT
GPIO 24: level=0 fsel=0 func=INPUT
GPIO 25: level=1 fsel=1 func=OUTPUT
GPIO 26: level=0 fsel=0 func=INPUT
GPIO 27: level=0 fsel=0 func=INPUT
BANK1 (GPIO 28 to 45):
GPIO 28: level=1 fsel=0 func=INPUT
GPIO 29: level=0 fsel=1 func=OUTPUT
GPIO 30: level=0 fsel=7 alt=3 func=CTS0
GPIO 31: level=0 fsel=7 alt=3 func=RTS0
GPIO 32: level=1 fsel=7 alt=3 func=TXD0
GPIO 33: level=1 fsel=7 alt=3 func=RXD0
GPIO 34: level=1 fsel=7 alt=3 func=SD1_CLK
GPIO 35: level=1 fsel=7 alt=3 func=SD1_CMD
GPIO 36: level=1 fsel=7 alt=3 func=SD1_DAT0
GPIO 37: level=1 fsel=7 alt=3 func=SD1_DAT1
GPIO 38: level=1 fsel=7 alt=3 func=SD1_DAT2
GPIO 39: level=1 fsel=7 alt=3 func=SD1_DAT3
GPIO 40: level=0 fsel=4 alt=0 func=PWM0
GPIO 41: level=0 fsel=4 alt=0 func=PWM1
GPIO 42: level=1 fsel=4 alt=0 func=GPCLK1
GPIO 43: level=1 fsel=4 alt=0 func=GPCLK2
GPIO 44: level=1 fsel=5 alt=1 func=SDA0
GPIO 45: level=1 fsel=5 alt=1 func=SCL0
BANK2 (GPIO 46 to 53):
GPIO 46: level=1 fsel=0 func=INPUT
GPIO 47: level=1 fsel=1 func=OUTPUT
GPIO 48: level=0 fsel=4 alt=0 func=SD0_CLK
GPIO 49: level=1 fsel=4 alt=0 func=SD0_CMD
GPIO 50: level=1 fsel=4 alt=0 func=SD0_DAT0
GPIO 51: level=1 fsel=4 alt=0 func=SD0_DAT1
GPIO 52: level=1 fsel=4 alt=0 func=SD0_DAT2
GPIO 53: level=1 fsel=4 alt=0 func=SD0_DAT3
```

lrnzo, 2019-08-05:
how important is it, that the output of `i2cdetect -y 1` is similar to the one pictured [here](https://github.com/ArduCAM/RaspberryPi/blob/master/data/check_msg.png)? back on raspbian buster i get this:
```
i2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- -- 
10: 10 -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- 64 -- -- -- -- -- -- -- -- -- -- -- 
70: 70 -- -- -- -- -- -- --                         
```
but it should be like the following, right?
```
i2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- 36 -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: 70 -- -- -- -- -- -- --                         
```
or why do these numbers  have a red frame around them in the mentioned picture?

lrnzo, 2019-08-05:
oh look, maybe i found something while playing around
```
i2cset -r 1 0x70 0x00 0x04
WARNING! This program can confuse your I2C bus, cause data loss and worse!
I will write to device file /dev/i2c-1, chip address 0x70, data address
0x00, data 0x04, mode byte.
Continue? [Y/n] 
Warning - data mismatch - wrote 0x04, read back 0xf0
```

lrnzo, 2019-08-06:
are there any datasheets for the board?

manoj027, 2019-08-07:
@UCTRONICS I am also using the same adapter which is older one according to you, and facing same issue, it seems I wasted my money, why arducam has not updated newer version on their cart, they should mention somewhere, can you tell me how I can use the older version with R-pi if I need to make changes in hardware I am ready.
It seems the new board is the same as the older one but they have placed another circuit on I2C, which is connected to I/P only, what is it some kind of MCU

lrnzo, 2019-08-07:
@manoj027 after having read your post i took a closer look at the pictures postet by @ahmed-shariff  and indeed there is an extra thingi on the "new" version of the board. finally i understand why i am confused. I still do not understand why the two different Boards have the same version (2.1) but at least i know that i got delivered the wrong part.

manoj027, 2019-08-08:
@ArduCAM @UCTRONICS Please can you reply, it's been only 3 days I have purchased this adapter, and it's not working, I can't spend more money on it, can you guys tell me how to fix this problem by hardware or by software anything.
Here are my observations:
1. When I connect camera normally in R-Pi 3B+, the camera works because it can detect video0 file,
2. When I connect multicamera adapter and reboot Pi video0 file is not creating so the camera is not running.
3. When I connect camera normally and reboot, it creates video0 file, then without shutting down Pi, I disconnect the camera and placed Multicamera adapter and connected 4 camera's to it, all 4 cameras are working because of video0 file has been already created, why the multicamera adapter is not creating video0 file?
@lrnzo you can also try, I think you will also get the same observation

lrnzo, 2019-08-08:
yes, i can reproduce that. i find it important to note that it works only if you allready have board connected to the gpios. so my procedure was:

1. connect single camera normally (-> `/dev/video0` should exist) and the multiplexer board to the gpio pins, not the CSI.
2. boot the pi
3. unplug the camera (it is easier to pull the cable out of the camera instead out of the pi)
4. plug the free end of the flex cable you just unplugged into the multiplexer board. it works. as far as i can tell by now. the i2cset error mentioned above still exists, but is maybe less severe than i thought

gerrievanzyl, 2019-10-21:
@UCTRONICS 
I am having this exact problem. I purchased the board from the Amazon link

Here is my output from the inti command

sudo ./init_camera.sh 
----Test i2c1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- 36 -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: 70 -- -- -- -- -- -- --                         
----Configure Adapter Board V2.1
Adapter Board V2.1 configure OK
----install v4l2 driver
v4l2 driver install OK
----Detecting the /dev/video0 device
Can't find video0 device, please check the hardware connection!

cjcbusatto, 2019-10-21:
> yes, i can reproduce that. i find it important to note that it works only if you allready have board connected to the gpios. so my procedure was:
> 
> 1. connect single camera normally and the multiplexer board to the gpio pins, not the CSI.
> 2. boot the pi and check that `/dev/video0` exists
> 3. unplug the camera (it is easier to pull the cable out of the camera instead out of the pi)
> 4. plug the free end of the flex cable you just unplugged into the multiplexer board. it works. as far as i can tell by now. the i2cset error mentioned above still exists, but is maybe less severe than i thought

@lrnzo I was able to make it work using the steps you have mentioned. Did you go further in the problem investigation? 
I logged the `udev` all events during the Raspberry boot and I could not find any message related to `video0` in the logs, it seems that the `/dev/video0` is not even tried to be created. My guess was that some errors were happening, but it seems to be an other problem.

cjcbusatto, 2019-11-17:
For those who are experiencing this issue:
- It seems to be a problem on the board we have. I contacted ArduCAM by e-mail and the boards I got directly from them worked without any problem.

Just for documentation, you can make the "not-working board" work, at least for development, following the [steps described by @lrnzo](https://github.com/ArduCAM/RaspberryPi/issues/11#issuecomment-519617533)

Our team tried to manually create the `/dev/video0`, in order to make the boards work without this trick, but we have not success so far. The interface is created, it passes on the ArduCAM's script test but it does not work to capture images.

jagan-veeraraghavan, 2019-12-14:
My init_camera.sh gives the following output where 36 is missing. 
This is with older V2.1 board along with Raspi 3+ B running Raspian V9.4
Any suggestions to fix this issue?

`----Test i2c1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: 70 -- -- -- -- -- -- --                         
----Configure Adapter Board V2.1
Adapter Board V2.1 configure OK
----install v4l2 driver
v4l2 driver install OK
----Detecting the /dev/video0 device
Can't find video0 device, please check the hardware connection! `

aylmerbritto, 2020-01-21:
You can refer this answer !! 
[Arducam-Multicam video0 interface](https://github.com/ArduCAM/RaspberryPi/issues/15#issuecomment-576517036)

ArduCAM, 2020-03-05:
Thank you so much aylmerbritto.

closed 2020-03-05 by ArduCAM