ArduCAM/RaspberryPi #14: sudo ./run_Arducam_Demo does nothing

GamerDerock, 2019-08-09:
I run the command: `sudo ./run_Arducam_Demo` and hit enter,
then it's just blank. Nothing appears.
`
pi@raspberry: ~/RaspberryPi/SPI_Camera $ sudo ./run_Arducam_Demo
      
`
Also will this work as a camera input? I'm trying to use this camera for an obstacle avoidance car

dlimoge, 2020-06-16:
I am having a similar issue. Was a solution ever offered?

Flap-Monster, 2020-11-24:
You have to change the 'peribase' to raspberry Pi 4. Take what I say with a pinch of sault but I believe the peribase is used to describe how the physical memory addresses and virtual memory addresses are translated via the arm CPU(I have come across it before when using DMA). The file that needs to be changed is this one in the /src directory bcm283x_board_driver.h. There is a #define which is commented out with //. Just comment out the non Pi 4 one by putting // in front of it and remove the // from the Pi4 specific one. Now I'm getting an SPI error but I know the camera is supported (been changing the gains/exposures etc using a software called raspicam c++.. specifically the opencv wrapper). Will update if I get past the SPI error!

ItzDerock, 2021-12-19:
> You have to change the 'peribase' to raspberry Pi 4. Take what I say with a pinch of sault but I believe the peribase is used to describe how the physical memory addresses and virtual memory addresses are translated via the arm CPU(I have come across it before when using DMA). The file that needs to be changed is this one in the /src directory bcm283x_board_driver.h. There is a #define which is commented out with //. Just comment out the non Pi 4 one by putting // in front of it and remove the // from the Pi4 specific one. Now I'm getting an SPI error but I know the camera is supported (been changing the gains/exposures etc using a software called raspicam c++.. specifically the opencv wrapper). Will update if I get past the SPI error! (Goes without saying you need to run make again to rebuild the .so)

@Flap-Monster Did you ever figure out the `SPI1 Interface Error!`? If not, did you find an alternative?