ArduCAM/RaspberryPi #15: V2.1 multicamera adapter issue

ajayarunachalam, 2019-08-30:
Hi folks,
I recently purchased the v2.1 multicamera adapter, and connected 4 pi cameras. On connecting as mentioned in the website, still can't find /dev/video0
Tried all the things mentioned like rpi-update, sudo modprobe bcm2835-v4l2, added it to /etc/modules. Also, installed v4l-utils etc. Check the cables too & tried replacing etc. None works.
On the other hand, if i connect the csi port to one of the cameras, and just plug the adapter on GPIO pins. It finds /dev/video0.

Any leads will be helpful. Thanks in advance

JojoDevel, 2019-09-08:
We had similar issues and solved them by rebooting the pi (sudo reboot) after executing the initCamera.sh script. Please do not do the reboot by unplugging the power as that makes a difference!

Does this work out for your?

ajayarunachalam, 2019-09-09:
Hi JojoDevel,
Thanks for the response. Following your suggestion, could find /dev/video0 , but the problem persists. On a 4 channel adapter board, only 2 camera's are being detected, while the other 2 are not. 
Also, tried changing the cables to re-verify, but didn't succeed.  Next, if i also restart the pi, can't find /dev/video0 (the same problem repeats)

UCTRONICS, 2019-09-10:
@ajayarunachalam @JojoDevel 
From the feedback of users, there are several reasons that lead to abnormal work.
Firstly, to ensure our test demo can work normally, you should ensure four cameras are all work normally. Because if one of them is defective, our test demo will not display.
So, we will advise users test each camera individually to ensure all of the four cameras can work normal. 
Then connect the four camera to the Adapter board, Notice, if you don't change our test code, you should connect the four camera to the Adapter board at the same time.
If not, the test demo will not display. 
In order to test each camera on the Adapter board individually, you should modify the test code. I advise you use this demo  I attached for you. Visit here to download https://github.com/ArduCAM/RaspberryPi/blob/master/Multi_Camera_Adapter/Multi_Adapter_Board_4Channel/Multi_Camera_Adapter_V2.1/AdapterTestDemo.py 
From the test demo, you will see , as default, we will test four cameras, so if you want to use two cameras, such as A and B, firstly, connect the camera to Adapter A and B channels. Then, you should comment the C and D camera in the code. If not, the system will hang up.

ajayarunachalam, 2019-09-10:
Hi UCTRONICS,
Thanks for the reply. I tested the 4 cameras individually connecting to CSI port, and all are working fine. On connecting them to adapter board, only 2 cameras are being functional, while all the 4 cameras connected to the board has to get connected.

Also, tested the AdapterTestDemo code, but still the same problem. Did you mean to test the cameras on adapter board individually one-by-one, or it is fine, the way I verified it by connecting to Raspberry Pi camera CSI port?

cjcbusatto, 2019-10-17:
I do have the same problem mentioned in this thread.

Is there a Raspberry Pi configuration which is tested and it is working? Like, Raspberry Pi 3 using jessie, etc?

cjcbusatto, 2019-11-17:
@ajayarunachalam I was experiencing the same issue as you. It seems to be a problem in the board I bought. I contacted ArduCAM by e-mail and the boards I got from them directly worked without any problem.

In any case, you can "make it work", at least for development, following the steps:
1. Connect a normal camera to the CSI Port
2. Once your Raspberry boot finishes, check for the `/dev/video0`, as you are using a working camera, it would be mounted without problems.
3. Replace the camera (with the Raspberry on) with the ArduCAM board, as the `video0` is already mounted, the ArduCAM will work without problems.

Our team tried to manually create the `/dev/video0`, in order to make the boards work without this trick, but we have not success so far. The interface is created, it passes on the ArduCAM's script test but it does not work to capture images.

aylmerbritto, 2020-01-21:
I have found something that could be of some help. As far as I have explored there is no way the multicam board can create a video0 interface because the camera is not exposed to the kernel modules until the i2c commands are set and multiplexed. So all you can do is run a shell script that could do the job for you and then use [raspiraw](https://github.com/6by9/raspiraw) to capture images and then play with Bayers color filter to get the proper image. 
1. `sudo modprobe i2c-dev`
2. `i2cset -y 1 0x70 0x00 0x04`
3.  This is for multiplexer to enable the camera A.
```
 sudo raspi-gpio set 4 op  # Enable Pin
 sudo raspi-gpio set 11 op # Select Pin 1
 sudo raspi-gpio set 18 op # Select Pin 2
 sudo rapi-gpio set 4 dl
 sudo rapi-gpio set 11 dl
 sudo rapi-gpio set 18 dh```

And then you could continue with [raspiraw](https://github.com/6by9/raspiraw) to capture images and then process it. Do let me know if you face any issues !! :)

closed 2020-03-05 by ArduCAM