ArduCAM/RaspberryPi #18: autofocus while recording video

cbasavaraj, 2019-11-12:
Hi, I've ordered the 8 MP auto-focus camera https://www.arducam.com/product/arducam-imx219-auto-focus-camera-module-drop-in-replacement-for-raspberry-pi-v2-and-nvidia-jetson-nano-camera/

A couple of questions while I wait for it:

- I've seen your autofocus.py script. While that sets the optimal focal length for a given view, do you know if it's possible to do this at regular intervals while recording to a stream as in: https://picamera.readthedocs.io/en/release-1.13/recipes1.html#recording-video-to-a-stream Maybe use the video splitter port?

- The first link says "The camera can be used on compute module CM3, CM3+, and NOT compatible with Raspberry Pi 3B/B+." Can you please confirm this? What about the Pi Zero?

Thanks!

capeieboi, 2021-11-19:
Hey! I have been using the autofocus camera with the pizero and it works fine, however I am having trouble integrating the streaming code with autofocus, what was your approach?

cbasavaraj, 2021-11-22:
Hey, I am not using autofocus anymore. But I think the general approach using the `picamera` library is to write to a buffer using the splitter port, and use that part to do autofocus or other stuff. See this example:
https://picamera.readthedocs.io/en/release-1.13/recipes2.html#recording-at-multiple-resolutions