ArduCAM/RaspberryPi #23: Can't see camera view while focusing

OrnithOrtion, 2020-01-15:
Hi there !
I have recently set up my new Camera Trap using Raspberry Pi and an ArduCam and wanted to focus on my Bird Feeder. 
I had trouble with _Motorized_Focus_Camera_Preview.py_. I couldn't view the camera stream while focusing. 
So I modified a bit the code to enable that :
`import os
import time
import sys
import threading
import pygame
import pygame.camera
from pygame.locals import *
from time import ctime, sleep

DEVICE = '/dev/video0'#the location of the camera
SIZE = (640, 480)#the size of the pygame window

pygame.init()
pygame.key.set_repeat(100)

def runFocus():
    temp_val = 512
    for event in pygame.event.get():
        if event.type ==KEYDOWN:
            print(temp_val)
            if event.key == K_UP:
                print('UP')
                if temp_val < 1000:
                    temp_val += 10
                else:
                    temp_val = temp_val
                value = (temp_val<<4) & 0x3ff0
                dat1 = (value>>8)&0x3f
                dat2 = value & 0xf0

                os.system("i2cset -y 0 0x0c %d %d" % (dat1,dat2))
            elif event.key==K_DOWN:
                print('DOWN')
                if temp_val <12 :
                    temp_val = temp_val
                else:
                    temp_val -= 10
                value = (temp_val<<4) & 0x3ff0
                dat1 = (value>>8)&0x3f
                dat2 = value & 0xf0
                os.system("i2cset -y 0 0x0c %d %d" % (dat1,dat2))

def runCamera():
    pygame.camera.init()
    display = pygame.display.set_mode(SIZE, 0)
    camera = pygame.camera.Camera(DEVICE, SIZE)
    camera.start()
    screen = pygame.surface.Surface(SIZE, 0, display)
    while True:
        screen = camera.get_image(screen)
        display.blit(screen, (0,0))
        pygame.display.flip()#only if the camera is upside down
        runFocus()
        #check if the user want to quit focusing 
        for event in pygame.event.get():
           if event.type == QUIT:
               camera.stop()
               pygame.quit()

if __name__ == "__main__":
   # t1 = threading.Thread(target=runFocus,args=("t1",))
   # t1.setDaemon(True)
   # t1.start()    
 runCamera()
`
I don't use threading anymore. It was not usefull. Moreover I couldn't use pygame key reading function while using it for camera preview.

I hope that comment will be usefull for guys who had the same issue, and perhaps for ArduCam itself.

abejarutas, 2020-09-15:
Hi samuel:
We have a similar problem, with the difference that I capture videos of birds with the camera.
Following your advice, I have downloaded
your code, but I still have the same problem, if I close the pygame window, the focus returns to the default position 512.
My camera is the official 8M raspberry camera with the arducam AF module, on a Raspberry Pi 3A +. With any of the Arducam AF demos, the same thing happens, you have to close them to apply raspivid or raspistill and the focus becomes useless. I think I'm doing something wrong, so can someone correct me, thank you.

OrnithOrtion, 2020-09-15:
Dear abejarutas,
I do not know what can be the issue. Maybe my version of the code do not include an important part of the Arducams's one, or maybe ArduCam camera did not think that it could be an issue ?
It should be a good Idea if you open a new Issue to indicate the problem to ArduCam team, and to renew the subject. It could be easier for you to find a solution.
ps. I'd be very excited to see how you take video of bird with Raspberry Pi, if you can show me how you achieve this, I'd be very happy, as I could not obtain any fluid movie while using `motion` software to detect their presence... (even with Raspi 4)
(If you want to see how I made my own system, you can go there : [WildLifePiCam](https://wildsamulus.toile-libre.org/diy/?article=picameratrap) [It is written in French, sorry...]

abejarutas, 2020-09-15:
Okay, I'll try to open a new issue,
P.S. I am trying to contact you through facebook and send you information to wildsamulus, but the contact form gives me an error:
"Fatal mistake
: Class 'PHPMailer \ PHPMailer' not found in
/data/web/2/1/wildsamulus.toile-libre.org/htdocs/contact.php
on line 20 "
I send you a link to some example videos, (I don't know if it's the right way on github. I'm sorry):
https://www.youtube.com/channel/UCcUkPa_-nwbI7nlPlUtYtVw/videos

abejarutas, 2020-09-18:
Hi Samuel:
While I wait to contact you, I am sending you a link to the program that works best for me:
https://mynaturewatch.net/daylight-camera-instructions
I have also opened a new issue and a thread in the arducam forum, to see if someone helps me.
#42 
Kind regards, waiting for your news:
abejarutas * gmail.com