ArduCAM/RaspberryPi #26: For those who are experiencing this issue:

asirajin, 2020-02-05:
For those who are experiencing this issue:
- It seems to be a problem on the board we have. I contacted ArduCAM by e-mail and the boards I got directly from them worked without any problem.

Just for documentation, you can make the "not-working board" work, at least for development, following the [steps described by @lrnzo](https://github.com/ArduCAM/RaspberryPi/issues/11#issuecomment-519617533)

Our team tried to manually create the `/dev/video0`, in order to make the boards work without this trick, but we have not success so far. The interface is created, it passes on the ArduCAM's script test but it does not work to capture images.

_Originally posted by @cjcbusatto in https://github.com/ArduCAM/RaspberryPi/issues/11#issuecomment-554735163_

asirajin, 2020-02-05:
I am having issue on this multicamera for raspberry pi. following the procedure for setting up 4 cameras on raspberry pi 3B+, when i run the command sudo ./init_camera.sh., it shows an error sudo: ./init_camera.sh: command not found. 
in the directory: /home/pi/RaspberryPi/Multi_Camera_Adapter/Multi_Adapter_Board_4Channel/Multi_Camera_Adapter_V2.1
the file init_camera.sh is there. please can somebody help on this

cjcbusatto, 2020-02-05:
Maybe you are already a super user in this machine, therefore using `sudo` causes the problem. 
Can you try to run the following commands and print the output?
```sh
chmod +x init_camera.sh
./init_camera.sh
```