ArduCAM/RaspberryPi #27: The switch between camera A and C get write failed error in terminal

roxanaluca, 2020-02-07:
Hi,
I have tried to use VideoCapture class from OpenCV library to get the image between camera A and C simultaneous. The problem is that I could not get the right frame per second and the images from each camera haven't any noise. The write failed error appears after 1s after running the program.
Multi camera adapter Arducam V2.1

```
import cv2
import os
import RPi.GPIO as gp

class Camera:
   def setup(self,id):
      if id == 0:
          i2c = "i2cset -y 1 0x70 0x00 0x04"
          os.system(i2c)
          gp.output(7, False)
          gp.output(11, False)
          gp.output(12, True)
      elif id == 1:
          i2c = "i2cset -y 1 0x70 0x00 0x05"
          os.system(i2c)
          gp.output(7, True)
          gp.output(11, False)
          gp.output(12, True)
      elif id == 2:
          i2c = "i2cset -y 1 0x70 0x00 0x06"
          os.system(i2c)
          gp.output(7, False)
          gp.output(11, True)
          gp.output(12, False)
      elif id == 3:
          i2c = "i2cset -y 1 0x70 0x00 0x07"
          os.system(i2c)
          gp.output(7, True)
          gp.output(11, True)
          gp.output(12, False)
    
   def __init__(self):
        
      self.vc = cv2.VideoCapture(0)
      self.vc.set(cv2.CAP_PROP_FRAME_HEIGHT,300)
      self.vc.set(cv2.CAP_PROP_FRAME_WIDTH,300)
      self.vc.set(cv2.CAP_PROP_FPS, 60)
    
      self.setup(0)
      self.cameraid = 0
      if self.vc.isOpened(): # try to get the first frame
         self.rval, self.frame = self.vc.read()
      else:
         self.rval = False
   
   def take_picture(self, index):
       if self.cameraid != index:
          self.setup(index)
          self.cameraid = index
       if self.rval:
          self.rval, self.frame = self.vc.read()
          return self.frame
       else:
          return None

   def __del__(self):
      self.vc.release()

```

Please tell me what can I do.
Many thanks,
Maria

UCTRONICS, 2020-02-11:
Hello,
Sorry for my later reply  due to my New Year's holiday.
You code exist problem. Due to we add i2c adapter, so you should init each camera. You should choose each channel and use  self.vc = cv2.VideoCapture(0) to init your camera. You can refer to our demo here https://github.com/ArduCAM/RaspberryPi/blob/master/Multi_Camera_Adapter/Multi_Adapter_Board_4Channel/Multi_Camera_Adapter_V2.1/4cam_cv2.py
What's more, we have move our support to www.arducam/forums. Let me know if you need more help.

roxanaluca, 2020-05-06:
I have changed completely the code for camera but I gave me the same error Write Failed when I run for many times. At one run of the program it works but for the next runs the error appears.
https://github.com/roxanaluca/Final-Project/blob/master/camera.py