ArduCAM/RaspberryPi #29: Multi Camera V2 Demos Don't Use Camera B

HCastano, 2020-02-25:
Hi, I have the  "Arducam Multi Camera Adapter Doubleplexer Stereo Module V2" connected to a Raspberry Pi Zero W and have been trying to get the `shell` samples to work. Unfortunately, they only seem to take photos from one camera (Camera A). The Stereo Module has `UC444 Rev-C` on the PCB.

I've tried the scripts in both `RaspberryPi/.../Multi_Adapter_Board_2Channel` and `RaspberryPi/.../Multi_Adapter_Board_2Channel_uc444` and haven't been able to get images from both cameras on either.

The script in `RaspberryPi/.../Multi_Adapter_Board_2Channel` has no errors/warnings. When I run it I see the camera LEDs toggling on when it takes a photo with Camera A, and turning off when Camera B is supposed to take a photo (Camera A just ends up taking another photo).

For the one in `RaspberryPi/.../Multi_Adapter_Board_2Channel_uc444` I get an error when trying to use `i2cset`, but I believe that's the same as #1 (just noting it down in case I'm mistaken).

I've tried flipping the cameras (e.g moving Camera A to the Camera B slot) and running the scripts again, and Camera A is still the only one that ends up taking a photo.

Let me know if you need more info.

UCTRONICS, 2020-02-26:
Don‘t worry and I will try my best to help you.
Please attach me your hardware connection and I will help you check it in detail.
It seems the GPIO4 does not be driven normally.
[image]

HCastano, 2020-02-26:
Here is the hardware setup @UCTRONICS.

[image]

[image]

In case it matters, I'm using the [following cameras](https://www.adafruit.com/product/1937).

RapZamora, 2020-02-27:
Same issue but I am using Raspberry Pi 3B+

HCastano, 2020-02-27:
After re-soldering the header onto the Raspberry Pi both cameras are taking photos using the `pi_cam_uc444` shell script. 

Immediately after re-soldering I wasn't able to get `i2c-tools` to work because of permission issues - they require `sudo` to use out of the box. Without the `i2cset` command running the cameras won't switch, so that's kind of an important detail. The error message I got in the case was:

`Camera control callback  cmd=0x4f525245mmal: No data received from sensor. Check all connections, including the Sunny one on the camera board`

Hope that helps others.

closed 2020-02-27 by HCastano

WhaSukGO, 2020-08-25:
> Same issue but I am using Raspberry Pi 3B+. I urgently need a solution for this because we purchased one 4 channel and 2 channel adapters, and both are not working properly @UCTRONICS

Any update? I'm currently in the same issue. Trying to get it work on my Rasberry Pi 3B