ArduCAM/RaspberryPi #32: Possible conflict with RaspberryPi display?

johnb8, 2020-03-22:
When using an arducam imx219 motorized focus camera with the RaspberryPi touch screen attached, the display and camera will occasionally lock up when taking pictures and changing focus. 

I'm running `Autofocus.py`.

I think there's some sort of i2c conflict that causes i2c-0 to lock up sometimes, resulting in the pi being unable to receive touch input from the display or control the camera. I can unlock it by running `rmmod i2c_dev` then `modprobe i2c_dev` then `i2cdetect -y 0`.

ArduCAM, 2020-03-23:
Yes, you are right, it is a known issue. The Raspberry pi touch screen use the I2C-0 as well, and once you touch the screen, there are communications on the bus, so it will conflict with the autofocus driver controller that use the same I2C-0 bus.

ArduCAM, 2020-03-23:
You probably need to use another Raspberry pi touch screen like: https://www.amazon.com/UCTRONICS-Raspberry-1024×600-Capacitive-Touchscreen/dp/B07VWDDWQ9

tkircher, 2020-03-29:
Is it possible to have a version of the library that uses i2c-1? I want to use both the official touchscreen and the arducam camera, and I'm willing to cut the SCL and SDA traces on the cable and connect them to the RPi i2c-1 pins (27, 28).

UCTRONICS, 2020-03-30:
Arducam release software i2c library. You can get it here: https://github.com/ArduCAM/MIPI_Camera/blob/sotwarei2c/RPI/lib/libarducam_mipicamera.so
It will not occupied i2c0 source.