ArduCAM/RaspberryPi #33: Multi-camera Adapter B0120 V2.2 + RPi 4B - All ports disabled executing init_camera.sh

ZannLii2, 2020-04-18:
Right on the get-go after purchasing ArduCAM 4-Channel Multi-camera Adapter B0120 V2.2, tried executing the camera modules, RPi 4B on Raspbian Buster seem to have crashed and disabled all of the ports including HDMI ports when trying to execute "init_camera.sh" shell file. Attempt to execute "4cam_cv3.py" doesn't display my 2 camera's outputs.
I'm not sure how to go from here after this. I hope you guys would help out about this issue.
Here are the details below:

Setup:
RPi 4B <-- SDCard 32GB <-- Raspbian Buster
           <-- HDMI <-- TV (VIZIO E321VL-TW)
           <-- LAN <-- Router
           <-- USB <-- USB Wireless Keyboard+Mouse combo
           <-- CSI <-- ArduCAM B0120 V2.2 <-- Camera A - RPi Camera V1.3
                                                                  <-- Camera C - RPi Camera V1.3

Terminal:
$ sudo raspi-config
   -- Enabled "Camera"
   -- Enabled "I2C"
$ sudo apt install python-pyqt5
$ sudo apt install python-opencv
$ git clone https://github.com/ArduCAM/RaspberryPi.git
$ cd RaspberryPi/Multi_Camera_Adapter/Multi_Adapter_Board_4Channel/Multi_Camera_Adapter_V2.2
$ sudo chmod +x init_camera.sh
$ sudo ./init_camera.sh

Status: CRASH
- half-way through Terminal execution, when "Start testing   the camera A", Rpi 4B seemingly crashed.
  | - ArduCAM Hat status LED lights up   OK.
  | - VNC got cut off 
  | - RPi 4B LAN Port seemingly disabled   (LAN Tx lights went out)
  | - HDMI TV output got cut off. No   signal detected.
  | - Keyboard interrupt   <Ctrl>+<C> doesn’t seem to respond.

Terminal output before CRASH:
     0  1   2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:             -- -- -- -- -- -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- 36 -- -- -- -- -- -- -- -- --
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
70: 70 -- -- -- -- -- -- --
---- Configure Adapter Board V2.1
Adapter Board V2.1 configure OK
---- install v4l2 driver configure
v4l2 driver install OK
---- detecting the dev/video0 device
video0 detected!
---- Start testing each camera
if all of them are normal, after it finished, you will see four image in the current path.
Start testing the camera A
####CRASH HAPPENS AFTER THIS LINE####

I've also attempted to troubleshoot by doing the following:
- add   dtparam=i2c_vc=on to /boot/config.txt,then reboot
- Terminal:
   $ i2cdetect -y 0
0    1  2  3    4  5  6    7  8  9    a  b  c    d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- --   --
10: -- -- -- -- -- -- -- -- -- -- -- --   -- -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- --   -- -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- --   -- -- -- --
40: -- -- -- -- -- -- -- -- -- -- -- --   -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- --   -- -- -- --
60: -- -- -- -- -- -- -- -- -- -- -- --   -- -- -- --
70: 70 -- -- -- -- -- -- --
- Run attempt using   ArduCAM's 4cam_cv3.py:
   $ python 4cam_cv3.py
- Output:
qt5ct: using qt5ct plugin
qt5ct: D-Bus global menu: no
VIDIOC_STREAMON: Invalid argument

hanifizzudinrahman, 2021-09-08:
me too facing the same issue as you, have you solve this problem?

ZannLii2, 2021-09-09:
no, haven't heard anything from the dev ever since. I've moved on to other projects using just 1 camera for now.

hanifizzudinrahman, 2021-09-09:
Ok, thx for the reply brother