ArduCAM/RaspberryPi #37: RPi Multi-Camera V2.1 - "init_camera.sh" appears to "freeze."

MattTucker22689, 2020-05-13:
So, I'm not 100% sure what's going on. "init_camera.sh" appears to "freeze."
The readout that I'm getting is:
> pi@raspberrypi:~/RaspberryPi/Multi_Camera_Adapter/Multi_Adapter_Board_4Channel/Multi_Camera_Adapter_V2.1 $ sudo ./init_camera.sh
----Test i2c1
Error: Could not open file `/dev/i2c-1' or `/dev/i2c/1': No such file or directory
----Configure Adapter Board V2.1
Error: Could not open file `/dev/i2c-1' or `/dev/i2c/1': No such file or directory
Adapter Board V2.1 configure OK
----install v4l2 driver
v4l2 driver install OK
----Detecting the /dev/video0 device
video0 detected!
---Start testing each camera
if all of them are normal, after it finished,you will see four image in the current path.
Start testing the camera A
Error: Could not open file `/dev/i2c-1' or `/dev/i2c/1': No such file or directory
Start testing the camera B
Error: Could not open file `/dev/i2c-1' or `/dev/i2c/1': No such file or directory
Start testing the camera C
Error: Could not open file `/dev/i2c-1' or `/dev/i2c/1': No such file or directory
Camera control callback  cmd=0x4f525245mmal: No data received from sensor. Check all connections, including the Sunny one on the camera board

This is the point that it consistently stops at.

I'm using an [RPi 4(4gig ram)](https://www.amazon.com/gp/product/B07TD42S27/ref=ppx_yo_dt_b_asin_title_o05_s01?ie=UTF8&psc=1), a [7" touch screen](https://www.amazon.com/gp/product/B07PV9BTJQ/ref=ppx_yo_dt_b_asin_title_o01_s00?ie=UTF8&psc=1), and two [8mp V2 cameras](https://www.amazon.com/gp/product/B01ER2SKFS/ref=ppx_yo_dt_b_asin_title_o05_s02?ie=UTF8&psc=1) with your [V2 MultiCam Adapter](https://www.amazon.com/gp/product/B07TYL36MC/ref=ppx_yo_dt_b_asin_title_o01_s01?ie=UTF8&psc=1).

And when I run "4cam_cv3.py" it just gives me 4 "blank" squares.

Here are the two images that it's taken. As you can see, they appear identical which they shouldn't be
[image]
[image]

ArduCAM, 2020-05-18:
You should enable the i2c1 firstly.
run sudo raspi-config 
[image]
[image]

MattTucker22689, 2020-05-22:
I did.

I'm reimaging the SD and going to try again from scratch.

MattTucker22689, 2020-06-13:
So... I have it sort of running. "init_camera.sh" works now. And it is depositing two images into the folder. 

[image]
[image]

But when I try to run "4cam_cv2.py" or "4cam_cv3.py" I only get four "empty" squares.

[image]

MattTucker22689, 2020-06-13:
I just bumped the GPU up to 256, and rebooted. 

Running "4cam_cv2.py" I now get this error:
[image]

Running "4cam_cv3.py" I get this:
[image]

And just to check if maybe I was missing a package or anything I tried reinstalling openCV:
[image]

I tried running "4cam_1.py" and "4cam_2.py" using Python 3 but it gave me a formatting error.

MattTucker22689, 2020-06-14:
So, after some digging... It appears the problem is with the version of OpenCV that installs now. 

The current default appears to be OpenCV 3.2.0. 
[image]

Which, apparently replaced cv2.cv with just cv2. 
(https://stackoverflow.com/questions/33177376/what-is-the-cv2-cv-replacement-in-opencv3/33199152)

But... when I made the replacement I instantly ran into a new problem...
[image]

Thoughts?

MattTucker22689, 2020-07-02:
After a LOT of hunting around.... this is what I've found:
https://github.com/skvark/opencv-python/issues/355

Thoughts????

@ArduCAM