ArduCAM/RaspberryPi #47: libarducam_vcm source code missing from repo

njbuch, 2021-03-21:
Hello. I cannot find the sourcecode for the libarducam_vcm library which is fundamental to the motorized focus camera. Would need that to move into long term support for our device.

bliptrip, 2021-07-09:
Hey njbuch.  I see that you didn't get help on this issue (or they sent you a private message).  Did you ever get source code for this?  If not, I'm not sure why they didn't put the code on github -- either they don't want to, or it's just a mistake and the support is aweful.

I found through an strace dump that they were essentially opening an fd on '/dev/i2c-10'.  In python, try this with 'ioctl' module (or write equivalent in C):
```
import os
import ioctl
import struct

fd = os.open('/dev/i2c-10', os.O_RDWR)

camdriver = ioctl.IOCTL.IO(b'\x07',0x6)
focus_val = 132

camdriver.perform(fd, 0xc) #Set i2c bus to force write as slave address (0x76) at 7-bit address 0xc.
os.write(fd, struct.pack('>H', focus_val << 4)) #Write focus value, but shift by 4 bytes to left and pack as 2-byte unsigned short in big-endian format
```

ArduCAM, 2021-07-13:
Hello,
This is Bin from arducam support. Terriblily sorry to reply you so late. 
In fact, About the so file, you can get the detail operations here https://github.com/ArduCAM/RaspberryPi/blob/ebbb0f988b44b267c6be0f5048132a91b5df86dd/Motorized_Focus_Camera/Autofocus.py . In order to be used by python code. we compile the C code  into a so file.
What's more, would you like to describe your issue in detail? I will try my best to help you solve all of your issue.
Feel free to contact support@arducam.com if you need more help.

surak, 2022-08-20:
The issue is that the SO file is 32 bit and one gets ```/usr/bin/ld: skipping incompatible /usr/lib/gcc/aarch64-linux-gnu/10/../../../../lib/libarducam_vcm.so when searching for -larducam_vcm``` when compiling it on a 64-bit system.

```bash
file /usr/lib/libarducam_vcm.so 
/usr/lib/libarducam_vcm.so: ELF 32-bit LSB shared object, ARM, EABI5 version 1 (SYSV), dynamically linked, BuildID[sha1]=14a85dadd7a15a60909de68eb04025eca2ffc4b8, not stripped
```