ArduCAM/RaspberryPi #50: Failed to set I2C address

ismyaki, 2021-04-23:
Why i runing sample but has error `Failed to set I2C address` 😱
```py
import cv2  # sudo apt-get install python-opencv
import numpy as py
import os
import time
from ctypes import *

# load arducam shared object file
arducam_vcm = CDLL('libs/libarducam_vcm.so')
try:
    import picamera
    from picamera.array import PiRGBArray
except:
    sys.exit(0)

def focusing(val):
    arducam_vcm.vcm_write(val)
    # print("focus value: {}".format(val))

def sobel(img):
    img_gray = cv2.cvtColor(img, cv2.COLOR_RGB2GRAY)
    img_sobel = cv2.Sobel(img_gray, cv2.CV_16U, 1, 1)
    return cv2.mean(img_sobel)[0]

def laplacian(img):
    img_gray = cv2.cvtColor(img, cv2.COLOR_RGB2GRAY)
    img_sobel = cv2.Laplacian(img_gray, cv2.CV_16U)
    return cv2.mean(img_sobel)[0]

def calculation(camera):
    rawCapture = PiRGBArray(camera)
    camera.capture(rawCapture, format="bgr", use_video_port=True)
    image = rawCapture.array
    rawCapture.truncate(0)
    return laplacian(image)

if __name__ == "__main__":
    # vcm init
    arducam_vcm.vcm_init()
    # open camera
    camera = picamera.PiCamera()

    # open camera preview
    camera.start_preview()
    # set camera resolution to 640x480(Small resolution for faster speeds.)
    camera.resolution = (640, 480)
    time.sleep(0.1)
    camera.shutter_speed = 30000
    print("Start focusing")

    max_index = 10
    max_value = 0.0
    last_value = 0.0
    dec_count = 0
    focal_distance = 10

    while True:
        # Adjust focus
        focusing(focal_distance)
        # Take image and calculate image clarity
        val = calculation(camera)
        # Find the maximum image clarity
        if val > max_value:
            max_index = focal_distance
            max_value = val

        # If the image clarity starts to decrease
        if val < last_value:
            dec_count += 1
        else:
            dec_count = 0
        # Image clarity is reduced by six consecutive frames
        if dec_count > 6:
            break
        last_value = val

        # Increase the focal distance
        focal_distance += 15
        if focal_distance > 1000:
            break

    # Adjust focus to the best
    focusing(max_index)
    time.sleep(1)
    # set camera resolution to 2592x1944
    camera.resolution = (1920, 1080)
    # save image to file.
    camera.capture("test.jpg")
    print("max index = %d,max value = %lf" % (max_index, max_value))
    # while True:
    #	time.sleep(1)

    camera.stop_preview()
    camera.close()
```

#### output 
```
pi@raspberrypi:~/camera_test $ sudo python3 test.py
Start focusing
Failed to set I2C address
Failed to set I2C address
Failed to set I2C address
Failed to set I2C address
Failed to set I2C address
Failed to set I2C address
Failed to set I2C address
Failed to set I2C address
Failed to set I2C address
Failed to set I2C address
Failed to set I2C address
Failed to set I2C address
Failed to set I2C address
Failed to set I2C address
max index = 55,max value = 3.607292
pi@raspberrypi:~/camera_test $
```

asnewman, 2021-05-07:
Looks like the most recent commit broke the code. I checked out 4d271ade72e1d9b0af889cd6351dafd460b10d00 and it worked.

ArduCAM, 2021-05-08:
@asnewman 
For the latest system version, the i2c adapter is i2c10 instead of i2c0. We have updated our lib to compatible with the latest system.
Please run **uname -a** command to check your kernel verson and run **ls /dev/i2c*** command to check if it exist i2c10 adapter.

asnewman, 2021-05-15:
@ArduCAM Thanks for the tip. When I ran the commands I get:
```
Linux raspberrypi 4.19.118-v7+ #1311 SMP Mon Apr 27 14:21:24 BST 2020 armv7l GNU/Linux
```
and
```
/dev/i2c-0  /dev/i2c-1
```

Do I need to run some sort of update command?

npaibrooklyn, 2021-07-01:
Any update on this? I am facing the same issue

bigdognj, 2021-11-14:
Just received my pi autofocus camera, installed everything as directed and am getting the same error as above when running the first Python demo. There are no I2C drivers in my /dev/ directory. What to do?

vidya0398, 2022-04-26:
I am also facing the same issue. Im getting error **failed to set i2c address** when I try to run Motorized_Focus_Camera_Preview.py and try to adjust the focus.

import os
import time
import sys
import _thread
from ctypes import *
keyboard = CDLL('./lib/libarducam_keyboard.so')
arducam_vcm =CDLL('./lib/libarducam_vcm.so')
UP      = 1
DOWN    = 2
SAVE    = 115
focus_val = 512;
step = 10
def run_camera(name):
    os.system("raspistill -t 0")
if __name__ == "__main__":
    _thread.start_new_thread(run_camera, ("run_camera",))
    #vcm init
    arducam_vcm.vcm_init()
   # camera = picamera.PiCamera()
   # camera.preview_fullscreen=False
   # preview=camera.start_preview()
    #set windows size
   # preview.window=(0,0,800,600)
    print("Please press up and down to adjust focus.")
    while True:
        keyVal = keyboard.processKeyEvent()
        time.sleep(0.01)
        if keyVal == UP:
            print("UP")
            focus_val = focus_val+step
            if  focus_val > 1023:
                focus_val = 1023
            print(focus_val)
            arducam_vcm.vcm_write(focus_val)
        if keyVal == DOWN:
            print("DOWN")
            focus_val = focus_val-step
            if focus_val < 0:
                focus_val = 0
            print(focus_val)
            arducam_vcm.vcm_write(focus_val)

This is the code I am using.

vidya0398, 2022-04-26:
[image]