ArduCAM/RaspberryPi #51: previewopencv for 2.2 multicamera doesnt work for python3 & C++

neherh, 2021-04-30:
i run make for 2.2 multi-camera and get this error when i make and run the previewopencv.

 ./previewOpencv 
Detected camera on channal:A
VIDEOIO ERROR: V4L2: Pixel format of incoming image is unsupported by OpenCV
ERROR! Unable to open camera

I attempted to do the python previewopencv using python3 and get this error:

[ WARN:0] global /tmp/pip-wheel-2l8ccy47/opencv-python/opencv/modules/videoio/src/cap_v4l.cpp (893) open VIDEOIO(V4L2:/dev/video0): can't open camera by index
Traceback (most recent call last):
  File "previewOpencv.py", line 8, in <module>
    Arducam_adapter_board.preview()
  File "/home/pi/Documents/code/RaspberryPi-master/Multi_Camera_Adapter/Multi_Adapter_Board_4Channel/Multi_Camera_Adapter_V2.2_python/AdapterBoard.py", line 81, in preview
    frame.dtype=np.uint8
AttributeError: 'NoneType' object has no attribute 'dtype'

neherh, 2021-07-28:
There were a few issues. I needed to double check my opencv installation. And on top of that, I needed to change the index in videocapture for it to work.

closed 2021-07-28 by neherh

GlShu7, 2021-12-23:
Hi, I'm using the multi-camera adapter module v2.2 and I'm trying to connect it to my RPi 4B, i followed the instructions from ArduCam's website but I got the exactly the same errors like you got and I can not take a photo/video with the cameras.
I use 3 rpi cameras v1.3 5MP. I installed before using this versions: python 3.9.6 and opencv 4.5.4

neherh, 2021-12-23:
@GlShu7 did you do exactly as I said previously? Did you reinstall opencv and check that installation? Then did you change the index in the videocapture function? Change it from -1 to anywhere between 0-25.

GlShu7, 2021-12-23:
Yes, but the index in the VideoCapture was (0).
I used the codes from the Arducam's github : https://github.com/ArduCAM/RaspberryPi/tree/master/Multi_Camera_Adapter/Multi_Adapter_Board_4Channel/Multi_Camera_Adapter_V2.2_python

neherh, 2021-12-23:
I would suggest that you change the index to another number that is NOT 0 or -1. 

I also reverted back to the old version that didn't include the .pyc files. But that was just a preference.

GlShu7, 2021-12-23:
I tried to change the number to 1 but still the same error

neherh, 2021-12-23:
continue every number to 25.

GlShu7, 2021-12-23:
I think the problem is in the line: frame.dtype=np.uint8 
isn't it ?

GlShu7, 2021-12-23:
I change it to 5 and I got another error, 
can't open camera b y index.

I use just 3 cameras and they are connected to A,C,D

neherh, 2021-12-23:
I am sorry. I think yours might be a different issue. I simply re-installed opencv and ensured v4l was working and then changed the index value. It worked otherwise; np.uint8 wasn't the issue. 

For the index, I literally kept incrementing the index until it worked. Some gave me errors, I ignored those and then went to another index.  I am not sure what your most recent error is related to.