ArduCAM/RaspberryPi #52: sh: 1: raspistill: not found

sohail759, 2021-05-31:
pi@raspberrypi:/RaspberryPi/Multi_Camera_Adapter/Multi_Adapter_Board_4Channel/Multi_Camera_Adapter_V2.1_C++ $ sudo ./init_camera.sh
----Test i2c1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- 0c -- -- -- 
10: 10 -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: 70 -- -- -- -- -- -- --                         
----Detecting the /dev/video0 device
video0 detected!
---Start testing each camera
if all of them are normal, after it finished,you will see four image in the current path.
Start testing the camera A
sh: 1: raspistill: not found
Start testing the camera B
sh: 1: raspistill: not found
Start testing the camera C
sh: 1: raspistill: not found
Start testing the camera D
sh: 1: raspistill: not found
pi@raspberrypi:/RaspberryPi/Multi_Camera_Adapter/Multi_Adapter_Board_4Channel/Multi_Camera_Adapter_V2.1_C++ $ sudo ./init_camera.sh
----Test i2c1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- 0c -- -- -- 
10: 10 -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: 70 -- -- -- -- -- -- --                         
----Detecting the /dev/video0 device
video0 detected!
---Start testing each camera
if all of them are normal, after it finished,you will see four image in the current path.
Start testing the camera A
sh: 1: raspistill: not found
Start testing the camera B
sh: 1: raspistill: not found
Start testing the camera C
sh: 1: raspistill: not found
Start testing the camera D

PLEASE HELP ME TO RESOLVE THIS ISSUE.

AllaboutJay, 2021-06-07:
Hello,

We suggest that:
1. Reinstall the image, the download link as follows:
https://downloads.raspberrypi.org/raspios_full_armhf/images/raspios_full_armhf-2021-05-28/2021-05-07-raspios-buster-armhf-full.zip

2. Then, follow Arducam DOC:
https://www.arducam.com/docs/cameras-for-raspberry-pi/multi-camera-adapter-board/multi-camera-adapter-board-v2-1/

hanifizzudinrahman, 2021-09-08:
this problem is same with me brother, have you solve this?

Shivanshu8211, 2024-06-09:
Hey, I booted my Raspberry Pi 4B on Ubuntu 20.04 and when I run the code, I'm getting a similar issue. If you have any solutions that I can run on Ubuntu, please let me know.